clash-lang/clash-prelude #17: Add Safe vector indexing

Ericson2314, 2015-05-13:
Just as we have pow2 block ram, etc, it would be nice if we had safe pow2 vector indexing.

I was just refactoring one of my processors to change the word size, and some bugs got caught with my tests instead of the type checker because I was using `!!`.

Maybe something like
```haskell
(!!!) :: Integral i
      => KnownNat (2 ^ BitSize i)
      => Vec (2 ^ BitSize i) a
      -> i -> a
(!!!) = (CL.!!)
```
should go in `CLaSH.Sized.Vector`?

Feel free to suggest a different identifier if you like the idea, of course.

christiaanb, 2015-05-17:
Perhaps call it `safeIndex`?
I mean... there are already:
[at](http://hackage.haskell.org/package/clash-prelude-0.7.5/docs/CLaSH-Sized-Vector.html#v:at)
[(!)](http://hackage.haskell.org/package/clash-prelude-0.7.5/docs/CLaSH-Prelude-BitIndex.html#v:-33-)
[(!!)](http://hackage.haskell.org/package/clash-prelude-0.7.5/docs/CLaSH-Sized-Vector.html#v:-33--33-)

I feel adding another indexing operator is going to make things more confusing, so I prefer a normal name.

Ericson2314, 2015-05-17:
Sure, that makes sense.

bgamari, 2016-07-28:
What remains to be done on this?

christiaanb, 2016-07-28:
Well, mostly we need to decide if we need more indexing operations than we already have for Vec: http://hackage.haskell.org/package/clash-prelude-0.10.10/docs/CLaSH-Sized-Vector.html#g:4, and then decide on a type signature.

As we currently have:

```haskell
(!!) :: (KnownNat n, Enum i) => Vec n a -> i -> a
at :: SNat m -> Vec (m + (n + 1)) a -> a
```