clash-lang/clash-prelude #27: GHC HEAD compile error

ggreif, 2015-09-28:
(I tried `--ghc-options="-fplugin GHC.TypeLits.Normalise"` but it did not help)

Maybe below error messages ring a bell what could have gone wrong...

```
[21 of 42] Compiling CLaSH.Sized.Fixed ( src/CLaSH/Sized/Fixed.hs, dist/dist-sandbox-fa345477/build/CLaSH/Sized/Fixed.o )

src/CLaSH/Sized/Fixed.hs:116:1: error:
    Couldn't match representation of type ?a2?
                             with that of ?rep (int + frac)?
      arising from a use of ?coerce?
    Relevant bindings include
      minBound :: Fixed rep int frac
        (bound at src/CLaSH/Sized/Fixed.hs:116:1)
    In the expression:
        coerce (minBound :: rep ((+) int frac)) :: Fixed rep int frac
    In an equation for ?minBound?:
        minBound
          = coerce (minBound :: rep ((+) int frac)) :: Fixed rep int frac
    When typechecking the code for ?minBound?
      in a derived instance for ?Bounded (Fixed rep int frac)?:
      To see the code I am typechecking, use -ddump-deriv
    In the instance declaration for ?Bounded (Fixed rep int frac)?

src/CLaSH/Sized/Fixed.hs:116:1: error:
    Couldn't match representation of type ?a3?
                             with that of ?rep (int + frac)?
      arising from a use of ?coerce?
    Relevant bindings include
      maxBound :: Fixed rep int frac
        (bound at src/CLaSH/Sized/Fixed.hs:116:1)
    In the expression:
        coerce (maxBound :: rep ((+) int frac)) :: Fixed rep int frac
    In an equation for ?maxBound?:
        maxBound
          = coerce (maxBound :: rep ((+) int frac)) :: Fixed rep int frac
    When typechecking the code for ?maxBound?
      in a derived instance for ?Bounded (Fixed rep int frac)?:
      To see the code I am typechecking, use -ddump-deriv
    In the instance declaration for ?Bounded (Fixed rep int frac)?

src/CLaSH/Sized/Fixed.hs:117:1: error:
    Couldn't match representation of type ?a1?
                             with that of ?rep (int + frac)?
      arising from a use of ?coerce?
    Relevant bindings include
      def :: Fixed rep int frac (bound at src/CLaSH/Sized/Fixed.hs:117:1)
    In the expression:
        coerce (def :: rep ((+) int frac)) :: Fixed rep int frac
    In an equation for ?def?:
        def = coerce (def :: rep ((+) int frac)) :: Fixed rep int frac
    When typechecking the code for ?def?
      in a derived instance for ?Default (Fixed rep int frac)?:
      To see the code I am typechecking, use -ddump-deriv
    In the instance declaration for ?Default (Fixed rep int frac)?

src/CLaSH/Sized/Fixed.hs:123:1: error:
    Couldn't match representation of type ?a0?
                             with that of ?rep (int + frac)?
      arising from a use of ?coerce?
    Relevant bindings include
      zeroBits :: Fixed rep int frac
        (bound at src/CLaSH/Sized/Fixed.hs:123:1)
    In the expression:
        coerce (zeroBits :: rep ((+) int frac)) :: Fixed rep int frac
    In an equation for ?zeroBits?:
        zeroBits
          = coerce (zeroBits :: rep ((+) int frac)) :: Fixed rep int frac
    When typechecking the code for ?zeroBits?
      in a derived instance for ?Bits (Fixed rep int frac)?:
      To see the code I am typechecking, use -ddump-deriv
    In the instance declaration for ?Bits (Fixed rep int frac)?
Failed to install clash-prelude-0.10
cabal.lnk: Error: some packages failed to install:
clash-prelude-0.10 failed during the building phase. The exception was:
ExitFailure 1
```

christiaanb, 2015-09-28:
Meh, this is just a GHC bug..., is there any difference between the `-ddump-deriv` of 7.10.* and HEAD?

christiaanb, 2015-09-28:
It seems that `coerce` is failing those type-classes where the members do _not_ have a function type, such as `Bounded`:

```haskell
class Bounded a where
  minBound :: a
  maxBound :: a
```

ggreif, 2015-09-28:
I do not have 7.10.2 around, but I am adding the HEAD (7.11.20150621) output.

```
==================== Derived instances ====================
Derived instances:
  instance Data.Bits.Bits
             (rep_azap (int_azaq GHC.TypeLits.+ frac_azar)) =>
           Data.Bits.Bits
             (CLaSH.Sized.Fixed.Fixed rep_azap int_azaq frac_azar) where
    (Data.Bits..&.)
      = GHC.Prim.coerce
          ((Data.Bits..&.) ::
             rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)
             -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)
                -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)) ::
          CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
          -> CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
             -> CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
    (Data.Bits..|.)
      = GHC.Prim.coerce
          ((Data.Bits..|.) ::
             rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)
             -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)
                -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)) ::
          CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
          -> CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
             -> CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
    Data.Bits.xor
      = GHC.Prim.coerce
          (Data.Bits.xor ::
             rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)
             -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)
                -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)) ::
          CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
          -> CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
             -> CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
    Data.Bits.complement
      = GHC.Prim.coerce
          (Data.Bits.complement ::
             rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)
             -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)) ::
          CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
          -> CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
    Data.Bits.shift
      = GHC.Prim.coerce
          (Data.Bits.shift ::
             rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)
             -> GHC.Types.Int
                -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)) ::
          CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
          -> GHC.Types.Int
             -> CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
    Data.Bits.rotate
      = GHC.Prim.coerce
          (Data.Bits.rotate ::
             rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)
             -> GHC.Types.Int
                -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)) ::
          CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
          -> GHC.Types.Int
             -> CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
    Data.Bits.zeroBits
      = GHC.Prim.coerce
          (Data.Bits.zeroBits ::
             rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)) ::
          CLaSH.Sized.Fixed.Fixed rep_ayOo int_ayOp frac_ayOq
    Data.Bits.bit
      = GHC.Prim.coerce
          (Data.Bits.bit ::
             GHC.Types.Int -> rep_ayOo ((GHC.TypeLits.+) int_ayOp frac_ayOq)) ::
[... 314 lines cut ...]
          ((GHC.Classes.==) ::
             rep_ayOJ ((GHC.TypeLits.+) int_ayOK frac_ayOL)
             -> rep_ayOJ ((GHC.TypeLits.+) int_ayOK frac_ayOL)
                -> GHC.Types.Bool) ::
          CLaSH.Sized.Fixed.Fixed rep_ayOJ int_ayOK frac_ayOL
          -> CLaSH.Sized.Fixed.Fixed rep_ayOJ int_ayOK frac_ayOL
             -> GHC.Types.Bool
    (GHC.Classes./=)
      = GHC.Prim.coerce
          ((GHC.Classes./=) ::
             rep_ayOJ ((GHC.TypeLits.+) int_ayOK frac_ayOL)
             -> rep_ayOJ ((GHC.TypeLits.+) int_ayOK frac_ayOL)
                -> GHC.Types.Bool) ::
          CLaSH.Sized.Fixed.Fixed rep_ayOJ int_ayOK frac_ayOL
          -> CLaSH.Sized.Fixed.Fixed rep_ayOJ int_ayOK frac_ayOL
             -> GHC.Types.Bool
  

Generic representation:
  
  Generated datatypes for meta-information:
  
  Representation types:

```

christiaanb, 2015-09-28:
Can you test if this works:

```haskell
{-# LANGUAGE ScopedTypeVariables, TypeOperators, KindSignatures, DataKinds #-}
module CoerceBug where

import Data.Coerce
import GHC.TypeLits

newtype  Fixed (rep :: Nat -> *) (int :: Nat) (frac :: Nat) = Fixed { unFixed :: rep (int + frac) }

minBoundTest :: forall rep int frac . Bounded (rep (int + frac))
             => Fixed rep int frac
minBoundTest = coerce (minBound :: rep (int + frac))
```

ggreif, 2015-09-28:
Yep, I get an error:

```
CoerceBug.hs:11:16: error:
    Couldn't match representation of type ?a0?
                             with that of ?rep (int + frac)?
      arising from a use of ?coerce?
    Relevant bindings include
      minBoundTest :: Fixed rep int frac (bound at CoerceBug.hs:11:1)
    In the expression: coerce (minBound :: rep (int + frac))
    In an equation for ?minBoundTest?:
        minBoundTest = coerce (minBound :: rep (int + frac))
```

christiaanb, 2015-09-28:
The `CoerceBug` module compiles just fine on GHC 7.10.2. Could you post a regression/bug report on the GHC trac?

ggreif, 2015-09-28:
Just tried `version 7.11.20150923`, it works there! Sorry for the noise, looks like I'll need an upgrade... :-(

christiaanb, 2015-09-28:
So this issue can be closed then?

ggreif, 2015-09-28:
Probably, but let me check. Tomorrow I'll have a freshly baked GHC. Then I'll try building CLaSH, and if all goes right I close this myself.

ggreif, 2015-09-29:
I am hanging in the prerequisite packages `th-desugar` & co., (which do not compile due to TH changes) but the original bug seems to be fixed in yesterday's GHC. So, closing.

closed 2015-09-29 by ggreif