clash-lang/clash-prelude #31: VHDL/Verilog probably shown wrong for clockWizard

ggreif, 2015-11-12:
Below this line

https://github.com/clash-lang/clash-prelude/blob/master/src/CLaSH/Annotations/TopEntity.hs#L357

I would not expect 'altpll' appearances. Tomorrow I'll check what is generated for xilinx.

ggreif, 2015-11-13:
Okay, after creating a "wrapper" for the ClockWizard generated IP in Vivado I get
```
architecture STRUCTURE of design_1 is
  component design_1_clk_wiz_0_0 is
  port (
    clk_in1_p : in STD_LOGIC;
    clk_in1_n : in STD_LOGIC;
    clk_out1 : out STD_LOGIC;
    reset : in STD_LOGIC;
    locked : out STD_LOGIC
  );
  end component design_1_clk_wiz_0_0;
```
@christiaanb Looks like it is in disconnect with what the `clockWizard` function is creating. Esp. it uses differential clock inputs. No idea how to proceed though. I am poking around a bit more. If you have any ideas/questions, please comment.

christiaanb, 2015-11-13:
Ah... I only ever tested with single-ended clock sources, never with differential clock sources.

This is an oversight on my part, and will be fixed in the next release of the prelude. Perhaps file a new issue that the `TopEntity` annotations cannot deal with differential clock sources?

ggreif, 2015-11-13:
@christiaanb good. this is now #33.

closed 2015-11-13 by ggreif