clash-lang/clash-prelude #48: Something triggering invalid infinite loop detection

polygonhell, 2016-03-06:
This occurs in the Haskell simulation, I can't easily verify the HDL version behaves the same currently. 

This might also be my code at fault, but I can't see how, it seems to be something introduced by either the lift into the Mealy state machine or Signal or the BlockRAM implementation, it's hard to pinpoint.

The code that triggers it is here https://github.com/polygonhell/J1LikeCPU

If I run "main", which just simulates the first 5 steps of the CPU, I get *** Exception: <<loop>>
If I run "test" which does basically the same thing, but without the Signal mechanics and using a List instead of RAM it appears to work.

What triggers it appears to be the next PC being dependent on the current instruction, if you goto line 193, and replace the code that generates pc' with simply pc' = pc + 1, main will complete correctly.
Note that in both cases pc' = pc' + 1 for all inputs.

Obviously instruction is dependent on the previous pc, since that's how it was looked up, but I can't see how this introduces an infinite loop.

christiaanb, 2016-03-07:
There is a combinational loop though in `system`
* `iin -> lowbit`
* `lowBit -> addr`
* `addr -> out`
* `out -> iin`

Then you might argue, that `evalM`, which is based on `eval` makes sure that the `iOutAddr` value of `out` is not combinationally dependent on `iin`, i.e. `iOutAddr` is controlled by a register, this is however not true:

In `eval`:
* `iOutAddr -> pc'`
* `pc' -> isImm`
* `isImm -> instruction`

And we found our combinational loop.

Replacing in `system`:

```haskell
iin = fn <$> iWord <*> lowBit
```

by:

```haskell
iin = fn <$> iWord <*> register 0 lowBit
```

ensures that we don't enter the `<loop>` exception, although the entire system doesn't work as it is supposed to anymore.

polygonhell, 2016-03-07:
Thanks, I completely missed that.

closed 2016-03-07 by polygonhell