clash-lang/clash-prelude #49: clockWizard broken

polygonhell, 2016-03-20:
Looks like it was broken by part of the differential signaling change
My top entity def looks like this

```
{-# ANN topEntity
  (TopEntity
    { t_name = "main"
    , t_inputs = []
    , t_outputs = ["LED", "TX"]
    , t_extraIn = [ ("SYS_CLK", 1)
                  ]
    , t_extraOut = []
    , t_clocks   = [ clockWizard "clkwiz50"
                             "SYS_CLK(0)"
                             "'0'" 
                   ]
}) #-}  
```
and it results in this
```
  clkwiz50_inst : entity clkwiz50
    port map
      (CLK_IN1 => SYS_CLK(0)
      ,CLK_IN1 => SYS_CLK(0)
      ,RESET   => '0'
      ,LOCKED  => clkwiz50_locked);
```
Note that CLK_IN1 is specifed twice, which causes XISE to give these errors

ERROR:HDLCompiler:433 - "\\psf\home\git\Clash\examples\Alu\vhdl\Main\main.vhdl" Line 25: Formal <clk_in1> is already associated.
ERROR:HDLCompiler:854 - "\\psf\home\git\Clash\examples\Alu\vhdl\Main\main.vhdl" Line 16: Unit <structural> ignored due to previous errors.

christiaanb, 2016-03-21:
Seems it was introduced by me when I implemented vivado-specific HDL output. I fixed it now, I'll make new release.

polygonhell, 2016-03-21:
cut and paste for the win.

christiaanb, 2016-03-21:
Fix is included in v0.6.16 of CLaSH: http://hackage.haskell.org/package/clash-ghc-0.6.16/changelog

closed 2016-03-21 by christiaanb