clash-lang/clash-prelude #50: (Feature req) Better optimisation for operator /=

ra1u, 2016-03-30:
Operator `/=` is not threated same at optimisation as `==`. Where there is possibility for code to be optimised away operator `==` works better (eliminates dead code) where `/=` does not

To show this we need some non trivial functions that generates files

common code

    module Pg2 where
    import CLaSH.Prelude

    type T = Signed 8

    funcA :: T -> T
    funcA a = fold (+) $ a :> 1 :> 2 :> 3 :> Nil

    funcB :: T -> T
    funcB a = fold (+) $ a :> 2 :> 3 :> 4 :> Nil

    topEntity = funcC d0

Using  `==` 

    funcC :: (KnownNat n) => SNat n -> T -> T
    funcC s
        | not ( snatToInteger s == 0 ) = funcA . funcA
        | otherwise = funcB . funcB

compiling to verilog generates this files

    Pg2_funcB_2.v
    Pg2_funcC_zdsfuncC_1.v
    Pg2_testbench.v
    Pg2_topEntity_0.v
    Pg2_topEntity.v

We see that `funcA` is optimised out
Now same code with operator `/=`

    funcC :: (KnownNat n) => SNat n -> T -> T
    funcC s
        | snatToInteger s /= 0 = funcA . funcA
        | otherwise = funcB . funcB

generated files

    Pg2_funcA_3.v
    Pg2_funcB_2.v
    Pg2_funcC_zdsfuncC_1.v
    Pg2_testbench.v
    Pg2_topEntity_0.v
    Pg2_topEntity.v

Here we see that `funcA` is also instantiated. It is also used in generated code, but I omitted this here.

ra1u, 2016-03-30:
Previous post was relevant with clash 6.10 , latest version 6.14 now yields error at synthesis with example using  `/=`

    Loading dependencies took 0.173811s 
    Applied 35 transformations
    Normalisation took 0.244344s
    *** Exception: CLaSH.Netlist(282): Not in normal form: Not a variable reference
    or primitive as subject of a case-statement

christiaanb, 2016-03-30:
I don't really know why versions beyond clash 0.6.10 trigger the above error, I'll see what I can do about it tomorrow. For now, I've implemented constant-reduction for `/=` on `Integer`, so that you get the optimised form, and no longer get the error. I'll also implement constant-reduction for other definitions of `/=` on the CLaSH primitive types.

closed 2016-04-07 by christiaanb