clash-lang/clash-prelude #54: Provide patterns for BitVector

ggreif, 2016-04-26:
I just came up with these bidirectional pattern synonyms for `high` and `low`:

``` haskell
pattern I <- ((==high) -> True)
  where I = high
pattern O <- ((==low) -> True)
  where O = low
```
I'll happily submit a pull request if you think that they are useful.

christiaanb, 2016-04-28:
Yeah... I don't know, you can already just pattern match on `0` and `1`. The only benefit of the pattern would be that it the argument you match on gets constrained to `Bit`, where matching on `0` or `1` only gets you a `(Num a, Eq a)` constraint.

Is that the purpose of the pattern synonym? better type inference?

ggreif, 2016-04-28:
Hey I did not know *that* trick. Cool! Closing...

closed 2016-04-28 by ggreif

christiaanb, 2016-04-28:
Yeah, `BitVector`, and hence `Bit` (`type Bit = BitVector 1`), has a instances of both `Num` of and `Eq`, so you can pattern match on integer literals.
Actually, `high` and `low` are sort-of legacy constants, left over from a time when `Bit` was defined as `data Bit = High | Low`.

I've actually thought about removing the `high` and `low` constants, given that you can alse just use `1` and `0`. In the end though, I decided not to, as removing those constants meant breaking code in the wild gratuitously. Perhaps I should reconsider that decision from a didactic viewpoint, where the availability of the `high` and `low` constants distracts from the fact that `Bit` has a `Num` instance.

ggreif, 2016-04-28:
You can deprecate them with a blurb to use 1 (resp 0 for low)

ggreif, 2016-04-28:
Should I submit a PR?

christiaanb, 2016-04-28:
It's ok, I'll add the deprecation message for the 0.11 release.