clash-lang/clash-prelude #57: Test suite failure with GHC 8

snoyberg, 2016-05-26:
This is a strange report, because the build log _looks_ like the test succeeded, but nonetheless it's reporting as a failure.

```
> /tmp/stackage-build9/clash-prelude-0.10.7$ runghc -clear-package-db -global-package-db -package-db=/home/stackage/work/builds/nightly/pkgdb Setup configure --enable-tests --package-db=clear --package-db=global --package-db=/home/stackage/work/builds/nightly/pkgdb --libdir=/home/stackage/work/builds/nightly/lib --bindir=/home/stackage/work/builds/nightly/bin --datadir=/home/stackage/work/builds/nightly/share --libexecdir=/home/stackage/work/builds/nightly/libexec --sysconfdir=/home/stackage/work/builds/nightly/etc --docdir=/home/stackage/work/builds/nightly/doc/clash-prelude-0.10.7 --htmldir=/home/stackage/work/builds/nightly/doc/clash-prelude-0.10.7 --haddockdir=/home/stackage/work/builds/nightly/doc/clash-prelude-0.10.7 --flags=
Configuring clash-prelude-0.10.7...
> /tmp/stackage-build9/clash-prelude-0.10.7$ runghc -clear-package-db -global-package-db -package-db=/home/stackage/work/builds/nightly/pkgdb Setup build
Building clash-prelude-0.10.7...
Preprocessing library clash-prelude-0.10.7...
[ 1 of 47] Compiling CLaSH.Sized.Internal.Index[boot] ( src/CLaSH/Sized/Internal/Index.hs-boot, dist/build/CLaSH/Sized/Internal/Index.o-boot )
[ 2 of 47] Compiling CLaSH.Sized.Internal.BitVector[boot] ( src/CLaSH/Sized/Internal/BitVector.hs-boot, dist/build/CLaSH/Sized/Internal/BitVector.o-boot )
[ 3 of 47] Compiling CLaSH.Sized.Vector[boot] ( src/CLaSH/Sized/Vector.hs-boot, dist/build/CLaSH/Sized/Vector.o-boot )
[ 4 of 47] Compiling CLaSH.Promoted.Symbol ( src/CLaSH/Promoted/Symbol.hs, dist/build/CLaSH/Promoted/Symbol.o )
[ 5 of 47] Compiling CLaSH.Promoted.Ord ( src/CLaSH/Promoted/Ord.hs, dist/build/CLaSH/Promoted/Ord.o )
[ 6 of 47] Compiling CLaSH.Promoted.Nat ( src/CLaSH/Promoted/Nat.hs, dist/build/CLaSH/Promoted/Nat.o )
[ 7 of 47] Compiling CLaSH.Promoted.Nat.TH ( src/CLaSH/Promoted/Nat/TH.hs, dist/build/CLaSH/Promoted/Nat/TH.o )
[ 8 of 47] Compiling CLaSH.Promoted.Nat.Literals ( src/CLaSH/Promoted/Nat/Literals.hs, dist/build/CLaSH/Promoted/Nat/Literals.o )
[ 9 of 47] Compiling CLaSH.Promoted.Nat.Unsafe ( src/CLaSH/Promoted/Nat/Unsafe.hs, dist/build/CLaSH/Promoted/Nat/Unsafe.o )
[10 of 47] Compiling CLaSH.Sized.RTree ( src/CLaSH/Sized/RTree.hs, dist/build/CLaSH/Sized/RTree.o )
[11 of 47] Compiling CLaSH.Class.Resize ( src/CLaSH/Class/Resize.hs, dist/build/CLaSH/Class/Resize.o )
[12 of 47] Compiling CLaSH.Class.Num  ( src/CLaSH/Class/Num.hs, dist/build/CLaSH/Class/Num.o )
[13 of 47] Compiling CLaSH.Signal.Internal ( src/CLaSH/Signal/Internal.hs, dist/build/CLaSH/Signal/Internal.o )
[14 of 47] Compiling CLaSH.Sized.Internal.BitVector ( src/CLaSH/Sized/Internal/BitVector.hs, dist/build/CLaSH/Sized/Internal/BitVector.o )
[15 of 47] Compiling CLaSH.Sized.BitVector ( src/CLaSH/Sized/BitVector.hs, dist/build/CLaSH/Sized/BitVector.o )
[16 of 47] Compiling CLaSH.Class.BitPack ( src/CLaSH/Class/BitPack.hs, dist/build/CLaSH/Class/BitPack.o )
[17 of 47] Compiling CLaSH.Prelude.BitIndex ( src/CLaSH/Prelude/BitIndex.hs, dist/build/CLaSH/Prelude/BitIndex.o )
[18 of 47] Compiling CLaSH.Prelude.BitReduction ( src/CLaSH/Prelude/BitReduction.hs, dist/build/CLaSH/Prelude/BitReduction.o )
[19 of 47] Compiling CLaSH.Sized.Internal.Index ( src/CLaSH/Sized/Internal/Index.hs, dist/build/CLaSH/Sized/Internal/Index.o )
[20 of 47] Compiling CLaSH.Sized.Index ( src/CLaSH/Sized/Index.hs, dist/build/CLaSH/Sized/Index.o )
[21 of 47] Compiling CLaSH.Sized.Vector ( src/CLaSH/Sized/Vector.hs, dist/build/CLaSH/Sized/Vector.o )

src/CLaSH/Sized/Vector.hs:1139:1: warning: [-Wredundant-constraints]
    • Redundant constraint: CmpNat (i + s) (s * n) ~ 'GT
    • In the type signature for:
           select :: CmpNat (i + s) (s * n) ~ 'GT =>
                     SNat f -> SNat s -> SNat n -> Vec (f + i) a -> Vec n a
[22 of 47] Compiling CLaSH.Sized.Internal.Signed ( src/CLaSH/Sized/Internal/Signed.hs, dist/build/CLaSH/Sized/Internal/Signed.o )
[23 of 47] Compiling CLaSH.Sized.Signed ( src/CLaSH/Sized/Signed.hs, dist/build/CLaSH/Sized/Signed.o )
[24 of 47] Compiling CLaSH.Sized.Internal.Unsigned ( src/CLaSH/Sized/Internal/Unsigned.hs, dist/build/CLaSH/Sized/Internal/Unsigned.o )
[25 of 47] Compiling CLaSH.Sized.Unsigned ( src/CLaSH/Sized/Unsigned.hs, dist/build/CLaSH/Sized/Unsigned.o )
[26 of 47] Compiling CLaSH.Sized.Fixed ( src/CLaSH/Sized/Fixed.hs, dist/build/CLaSH/Sized/Fixed.o )
[27 of 47] Compiling CLaSH.Signal.Bundle ( src/CLaSH/Signal/Bundle.hs, dist/build/CLaSH/Signal/Bundle.o )
[28 of 47] Compiling CLaSH.Signal.Explicit ( src/CLaSH/Signal/Explicit.hs, dist/build/CLaSH/Signal/Explicit.o )
[29 of 47] Compiling CLaSH.Signal     ( src/CLaSH/Signal.hs, dist/build/CLaSH/Signal.o )
[30 of 47] Compiling CLaSH.Prelude.BlockRam ( src/CLaSH/Prelude/BlockRam.hs, dist/build/CLaSH/Prelude/BlockRam.o )
[31 of 47] Compiling CLaSH.Prelude.BlockRam.File ( src/CLaSH/Prelude/BlockRam/File.hs, dist/build/CLaSH/Prelude/BlockRam/File.o )
[32 of 47] Compiling CLaSH.Prelude.Mealy ( src/CLaSH/Prelude/Mealy.hs, dist/build/CLaSH/Prelude/Mealy.o )
[33 of 47] Compiling CLaSH.Prelude.DataFlow ( src/CLaSH/Prelude/DataFlow.hs, dist/build/CLaSH/Prelude/DataFlow.o )

src/CLaSH/Prelude/DataFlow.hs:540:10: warning: [-Wredundant-constraints]
    • Redundant constraint: KnownNat m
    • In the instance declaration for ‘LockStep (Vec m en) (Vec m a)’
[34 of 47] Compiling CLaSH.Prelude.Moore ( src/CLaSH/Prelude/Moore.hs, dist/build/CLaSH/Prelude/Moore.o )
[35 of 47] Compiling CLaSH.Prelude.RAM ( src/CLaSH/Prelude/RAM.hs, dist/build/CLaSH/Prelude/RAM.o )
[36 of 47] Compiling CLaSH.Prelude.ROM ( src/CLaSH/Prelude/ROM.hs, dist/build/CLaSH/Prelude/ROM.o )
[37 of 47] Compiling CLaSH.Prelude.ROM.File ( src/CLaSH/Prelude/ROM/File.hs, dist/build/CLaSH/Prelude/ROM/File.o )
[38 of 47] Compiling CLaSH.Prelude.Synchronizer ( src/CLaSH/Prelude/Synchronizer.hs, dist/build/CLaSH/Prelude/Synchronizer.o )
[39 of 47] Compiling CLaSH.Prelude.Explicit.Safe ( src/CLaSH/Prelude/Explicit/Safe.hs, dist/build/CLaSH/Prelude/Explicit/Safe.o )
[40 of 47] Compiling CLaSH.Prelude.Testbench ( src/CLaSH/Prelude/Testbench.hs, dist/build/CLaSH/Prelude/Testbench.o )
[41 of 47] Compiling CLaSH.Prelude.Explicit ( src/CLaSH/Prelude/Explicit.hs, dist/build/CLaSH/Prelude/Explicit.o )
[42 of 47] Compiling CLaSH.Signal.Delayed ( src/CLaSH/Signal/Delayed.hs, dist/build/CLaSH/Signal/Delayed.o )
[43 of 47] Compiling CLaSH.Annotations.TopEntity ( src/CLaSH/Annotations/TopEntity.hs, dist/build/CLaSH/Annotations/TopEntity.o )
[44 of 47] Compiling CLaSH.Prelude.Safe ( src/CLaSH/Prelude/Safe.hs, dist/build/CLaSH/Prelude/Safe.o )
[45 of 47] Compiling CLaSH.Prelude    ( src/CLaSH/Prelude.hs, dist/build/CLaSH/Prelude.o )
[46 of 47] Compiling CLaSH.Tutorial   ( src/CLaSH/Tutorial.hs, dist/build/CLaSH/Tutorial.o )
[47 of 47] Compiling CLaSH.Examples   ( src/CLaSH/Examples.hs, dist/build/CLaSH/Examples.o )
Preprocessing test suite 'doctests' for clash-prelude-0.10.7...
[1 of 1] Compiling Main             ( tests/doctests.hs, dist/build/doctests/doctests-tmp/Main.o )
Linking dist/build/doctests/doctests ...
> /tmp/stackage-build9/clash-prelude-0.10.7$ dist/build/doctests/doctests
```

christiaanb, 2016-05-26:
Thanks for the report. I'll check it out on my own machine to see what is exactly going on.

christiaanb, 2016-05-27:
I think some sort of watchdog/timeout is triggered because the testsuite is going into an infinite loop on my machine.

The problem is, in my testsuite I have code that, on ghc 7.10 would result in an "Exception: <<loop>>" error, which I would catch. However, on ghc 8.0.1, it seems that this code no longer triggers the "Exception <<loop>>", but actually goes into an infinite loop...

bergmark, 2016-08-28:
New output, but still broken unfortunately

```
> /tmp/stackage-build8/clash-prelude-0.10.14$ runghc -clear-package-db -global-package-db -package-db=/var/stackage/work/builds/nightly/pkgdb Setup configure --enable-tests --package-db=clear --package-db=global --package-db=/var/stackage/work/builds/nightly/pkgdb --libdir=/var/stackage/work/builds/nightly/lib --bindir=/var/stackage/work/builds/nightly/bin --datadir=/var/stackage/work/builds/nightly/share --libexecdir=/var/stackage/work/builds/nightly/libexec --sysconfdir=/var/stackage/work/builds/nightly/etc --docdir=/var/stackage/work/builds/nightly/doc/clash-prelude-0.10.14 --htmldir=/var/stackage/work/builds/nightly/doc/clash-prelude-0.10.14 --haddockdir=/var/stackage/work/builds/nightly/doc/clash-prelude-0.10.14 --flags=
Configuring clash-prelude-0.10.14...
> /tmp/stackage-build8/clash-prelude-0.10.14$ runghc -clear-package-db -global-package-db -package-db=/var/stackage/work/builds/nightly/pkgdb Setup build
Building clash-prelude-0.10.14...
Preprocessing library clash-prelude-0.10.14...
[ 1 of 48] Compiling CLaSH.Sized.Internal.Index[boot] ( src/CLaSH/Sized/Internal/Index.hs-boot, dist/build/CLaSH/Sized/Internal/Index.o-boot )
[ 2 of 48] Compiling CLaSH.Sized.Internal.BitVector[boot] ( src/CLaSH/Sized/Internal/BitVector.hs-boot, dist/build/CLaSH/Sized/Internal/BitVector.o-boot )
[ 3 of 48] Compiling CLaSH.Sized.Vector[boot] ( src/CLaSH/Sized/Vector.hs-boot, dist/build/CLaSH/Sized/Vector.o-boot )
[ 4 of 48] Compiling CLaSH.Promoted.Symbol ( src/CLaSH/Promoted/Symbol.hs, dist/build/CLaSH/Promoted/Symbol.o )
[ 5 of 48] Compiling CLaSH.Promoted.Ord ( src/CLaSH/Promoted/Ord.hs, dist/build/CLaSH/Promoted/Ord.o )
[ 6 of 48] Compiling CLaSH.Promoted.Nat ( src/CLaSH/Promoted/Nat.hs, dist/build/CLaSH/Promoted/Nat.o )
[ 7 of 48] Compiling CLaSH.Promoted.Nat.TH ( src/CLaSH/Promoted/Nat/TH.hs, dist/build/CLaSH/Promoted/Nat/TH.o )
[ 8 of 48] Compiling CLaSH.Promoted.Nat.Literals ( src/CLaSH/Promoted/Nat/Literals.hs, dist/build/CLaSH/Promoted/Nat/Literals.o )
[ 9 of 48] Compiling CLaSH.Promoted.Nat.Unsafe ( src/CLaSH/Promoted/Nat/Unsafe.hs, dist/build/CLaSH/Promoted/Nat/Unsafe.o )
[10 of 48] Compiling CLaSH.Sized.RTree ( src/CLaSH/Sized/RTree.hs, dist/build/CLaSH/Sized/RTree.o )
[11 of 48] Compiling CLaSH.Class.Resize ( src/CLaSH/Class/Resize.hs, dist/build/CLaSH/Class/Resize.o )
[12 of 48] Compiling CLaSH.Class.Num  ( src/CLaSH/Class/Num.hs, dist/build/CLaSH/Class/Num.o )
[13 of 48] Compiling CLaSH.Signal.Internal ( src/CLaSH/Signal/Internal.hs, dist/build/CLaSH/Signal/Internal.o )
[14 of 48] Compiling CLaSH.Sized.Internal.BitVector ( src/CLaSH/Sized/Internal/BitVector.hs, dist/build/CLaSH/Sized/Internal/BitVector.o )
[15 of 48] Compiling CLaSH.Sized.BitVector ( src/CLaSH/Sized/BitVector.hs, dist/build/CLaSH/Sized/BitVector.o )
[16 of 48] Compiling CLaSH.Class.BitPack ( src/CLaSH/Class/BitPack.hs, dist/build/CLaSH/Class/BitPack.o )
[17 of 48] Compiling CLaSH.Prelude.BitIndex ( src/CLaSH/Prelude/BitIndex.hs, dist/build/CLaSH/Prelude/BitIndex.o )
[18 of 48] Compiling CLaSH.Prelude.BitReduction ( src/CLaSH/Prelude/BitReduction.hs, dist/build/CLaSH/Prelude/BitReduction.o )
[19 of 48] Compiling CLaSH.Sized.Internal.Index ( src/CLaSH/Sized/Internal/Index.hs, dist/build/CLaSH/Sized/Internal/Index.o )
[20 of 48] Compiling CLaSH.Sized.Index ( src/CLaSH/Sized/Index.hs, dist/build/CLaSH/Sized/Index.o )
[21 of 48] Compiling CLaSH.Sized.Vector ( src/CLaSH/Sized/Vector.hs, dist/build/CLaSH/Sized/Vector.o )

src/CLaSH/Sized/Vector.hs:1145:1: warning: [-Wredundant-constraints]
    • Redundant constraint: CmpNat (i + s) (s * n) ~ 'GT
    • In the type signature for:
           select :: CmpNat (i + s) (s * n) ~ 'GT =>
                     SNat f -> SNat s -> SNat n -> Vec (f + i) a -> Vec n a
[22 of 48] Compiling CLaSH.Sized.Internal.Signed ( src/CLaSH/Sized/Internal/Signed.hs, dist/build/CLaSH/Sized/Internal/Signed.o )
[23 of 48] Compiling CLaSH.Sized.Signed ( src/CLaSH/Sized/Signed.hs, dist/build/CLaSH/Sized/Signed.o )
[24 of 48] Compiling CLaSH.Sized.Internal.Unsigned ( src/CLaSH/Sized/Internal/Unsigned.hs, dist/build/CLaSH/Sized/Internal/Unsigned.o )
[25 of 48] Compiling CLaSH.Sized.Unsigned ( src/CLaSH/Sized/Unsigned.hs, dist/build/CLaSH/Sized/Unsigned.o )
[26 of 48] Compiling CLaSH.Sized.Fixed ( src/CLaSH/Sized/Fixed.hs, dist/build/CLaSH/Sized/Fixed.o )
[27 of 48] Compiling CLaSH.Signal.Bundle ( src/CLaSH/Signal/Bundle.hs, dist/build/CLaSH/Signal/Bundle.o )
[28 of 48] Compiling CLaSH.Signal.Explicit ( src/CLaSH/Signal/Explicit.hs, dist/build/CLaSH/Signal/Explicit.o )
[29 of 48] Compiling CLaSH.Signal     ( src/CLaSH/Signal.hs, dist/build/CLaSH/Signal.o )
[30 of 48] Compiling CLaSH.Prelude.BlockRam ( src/CLaSH/Prelude/BlockRam.hs, dist/build/CLaSH/Prelude/BlockRam.o )
[31 of 48] Compiling CLaSH.Prelude.BlockRam.File ( src/CLaSH/Prelude/BlockRam/File.hs, dist/build/CLaSH/Prelude/BlockRam/File.o )
[32 of 48] Compiling CLaSH.Prelude.Mealy ( src/CLaSH/Prelude/Mealy.hs, dist/build/CLaSH/Prelude/Mealy.o )
[33 of 48] Compiling CLaSH.Prelude.DataFlow ( src/CLaSH/Prelude/DataFlow.hs, dist/build/CLaSH/Prelude/DataFlow.o )

src/CLaSH/Prelude/DataFlow.hs:540:10: warning: [-Wredundant-constraints]
    • Redundant constraint: KnownNat m
    • In the instance declaration for ‘LockStep (Vec m en) (Vec m a)’
[34 of 48] Compiling CLaSH.Prelude.Moore ( src/CLaSH/Prelude/Moore.hs, dist/build/CLaSH/Prelude/Moore.o )
[... cut ...]
              • In the expression: setSlice d6 d5 0 (- 5 :: Signed 6)
                In an equation for ‘it’: it = setSlice d6 d5 0 (- 5 :: Signed 6)
### Failure in src/CLaSH/Signal/Internal.hs:691: expression `simulate (register 8) [1, 2, 3]'
expected: [8,1,2,3...
 but got: [8,1,2,3*** Exception: finite list
          CallStack (from HasCallStack):
            error, called at src/CLaSH/Signal/Internal.hs:684:32 in main:CLaSH.Signal.Internal
Examples: 517  Tried: 507  Errors: 0  Failures: 32
```

christiaanb, 2017-01-29:
Fixed with the release of 0.11

closed 2017-01-29 by christiaanb