clash-lang/clash-prelude #66: 0.10.13: Conflicting exports for ‘Min’ and ‘Max’

juhp, 2016-08-20:
I don't understand why this started happening in the Stackage Nightly build, but reporting it here:

```
[45 of 48] Compiling CLaSH.Prelude.Safe ( src/CLaSH/Prelude/Safe.hs, dist/build/CLaSH/Prelude/Safe.o )

src/CLaSH/Prelude/Safe.hs:93:5: error:
    Conflicting exports for ‘Min’:
       ‘module GHC.TypeLits.Extra’ exports ‘GHC.TypeLits.Extra.Min’
         imported from ‘GHC.TypeLits.Extra’ at src/CLaSH/Prelude/Safe.hs:111:1-25
       ‘module CLaSH.Promoted.Ord’ exports ‘CLaSH.Promoted.Ord.Min’
         imported from ‘CLaSH.Promoted.Ord’ at src/CLaSH/Prelude/Safe.hs:135:1-25
         (and originally defined at src/CLaSH/Promoted/Ord.hs:22:1-44)

src/CLaSH/Prelude/Safe.hs:93:5: error:
    Conflicting exports for ‘Max’:
       ‘module GHC.TypeLits.Extra’ exports ‘GHC.TypeLits.Extra.Max’
         imported from ‘GHC.TypeLits.Extra’ at src/CLaSH/Prelude/Safe.hs:111:1-25
       ‘module CLaSH.Promoted.Ord’ exports ‘CLaSH.Promoted.Ord.Max’
         imported from ‘CLaSH.Promoted.Ord’ at src/CLaSH/Prelude/Safe.hs:135:1-25
         (and originally defined at src/CLaSH/Promoted/Ord.hs:27:1-44)
```

I could not reproduce locally with ``stack -resolver nightly build clash-prelude.

christiaanb, 2016-08-20:
Yes, this is my fault, I (erroneously) updated the upper bound for the `ghc-typelits-extra` dependency on hackage. I've reverted this now.

I'll release a new version of clash-prelude, one that that supports `ghc-typelits-extra-0.2` soon.

juhp, 2016-08-20:
Okay thanks, I downgraded ghc-typelits-extra for now in Stackage Nightly.

christiaanb, 2016-08-21:
I have just released `clash-prelude-0.10.14`, which successfully builds against `ghc-typelits-extra-0.2`, on hackage: http://hackage.haskell.org/package/clash-prelude-0.10.14

commit 9dd9dd43 (2016-08-21): Build against ghc-typelits-extra-0.2

Fixes #66
commit 3dd098b5 (2016-08-22): Use `Max` from `ghc-typelits-extra`

Fixes #66
closed 2016-08-21 by christiaanb