clash-lang/clash-prelude #67: maxIndex on Vec is inconsistent with maxIndex from Data.Vector

tomjaguarpaw, 2016-08-27:
[Data.Vector](https://hackage.haskell.org/package/vector-0.11.0.0/docs/Data-Vector.html) `maxIndex` returns "the index of the maximum element of the vector".  This is completely different from Clash's `maxIndex` from [CLaSH.Sized.Vector](https://hackage.haskell.org/package/clash-prelude-0.10.7/docs/CLaSH-Sized-Vector.html#v:maxIndex) which is "The index (subscript) of the last element in a Vector as an Int value.".

At the very least there should be a big warning that these are different.  However, Clash's `maxIndex` is redundant because that information is present in the type so I would go even further and either remove `maxIndex` from Clash or (more dangerously) change it to the `Data.Vector` behavior.

christiaanb, 2016-08-28:
I would prefer removing it then. 

`maxIndex` is a function I added to the `clash-prelude` when we build the first prototype of CLaSH in 2009, at a time I didn't even know about `Data.Vector`. I think I'll deprecate `maxIndex` in clash-prelude-0.7, and then remove it in clash-prelude-1.0. I just don't know what to put exactly in the deprecation message?

`"'maxIndex' will be removed in CLaSH 1.0, use 'length v - 1' instead."` ?

tomjaguarpaw, 2016-08-28:
That sounds fine.

commit b241e1a9 (2016-09-01): Deprecate `maxIndex`

Fixes #67
closed 2016-09-01 by christiaanb