clash-lang/clash-prelude #70: Better interface for regEn

tomjaguarpaw, 2016-08-29:
`regEn` would be strictly typesafer if its signature were

    regEn :: a -> Signal (Maybe a) -> Signal a

rather than

    regEn :: a -> Signal Bool -> Signal a -> Signal a

This is very similar to https://github.com/clash-lang/clash-prelude/issues/42

tomjaguarpaw, 2016-08-29:
I don't suggest to break this type signature but the new version could be added and the old one deprecated eventually.

tomjaguarpaw, 2016-08-29:
````haskell
regEnMaybe :: a -> Signal' clk (Maybe a) -> Signal' clk a
regEnMaybe = regEnMaybe' systemClock

regEnMaybe' :: SClock clk -> a -> Signal' clk (Maybe a) -> Signal' clk a
regEnMaybe' initial clk input = regEn' clk initial (fmap isJust input)
                                   $ fmap fromJust input
````

tomjaguarpaw, 2016-08-29:
````haskell
regEnMaybe :: a -> Signal (Maybe a) -> Signal a
regEnMaybe = regEnMaybe' systemClock

regEnMaybe' :: SClock clk -> a -> Signal' clk (Maybe a) -> Signal' clk a
regEnMaybe' clk initial input = regEn' clk initial (fmap isJust input) $ fmap fromJust input
````

christiaanb, 2016-08-31:
I prefer the name `regMaybe`, thoughts?

tomjaguarpaw, 2016-08-31:
`regMaybe` seems good.

tomjaguarpaw, 2016-08-31:
In fact `registerMaybe` is even better.

commit 337c6912 (2016-08-31): Add `registerMaybe`

* Thanks to @tomjaguarpaw.
* Deprecates `regEn` in favour of `registerMaybe`.
* Closes #70
closed 2016-08-31 by christiaanb