clash-lang/clash-prelude #72: Suggestions for some utility functions

comonoidial, 2016-09-02:
Looking at some clash code I found useful functions missing from the prelude.

* snatToNum (as Integer is rarely used in designs)
* bitCast = unpack . pack  (implemented with zero cost in hardware)
* constructing a vector from a index mapping function  `(Index n -> a) -> Vec n a`
* multiplexor function with n inputs on Signal types so that it be conveniently written on one line (using a list of inputs or variadic function, or whatever synthesizes efficiently) instead the now common:
```haskell
res = myMux <$> sels <*> xs <*> ys <*> zs
  where myMux A _ _ _ = 0
            myMux B x _ _ = x
            myMux C _ y _ = y
            myMux D _ _ z = z
-- want to write something like:
multiplexor sels (A,0) (B,xs) (C,ys) (D,zs)
```

christiaanb, 2016-09-02:
There is `bitCoerce`: http://hackage.haskell.org/package/clash-prelude-0.10.14/docs/CLaSH-Class-BitPack.html#v:bitCoerce

christiaanb, 2016-09-02:
http://hackage.haskell.org/package/clash-prelude-0.10.14/docs/CLaSH-Sized-Vector.html#v:backpermute is isomorphic to `(Index n -> a) -> Vec n a`.

christiaanb, 2016-09-02:
Would:

```haskell
multiplexor :: (KnownNat n, Enum sel) => Signal sel -> Vec n (Signal a) -> Signal a
multiplexor sel alts = (!!) <$> bundle alts <*> sel
```

work for you?

so you can write:

```haskell
multiplexor sels (0 :> xs :> ys :> zs :> Nil)
```

ggreif, 2016-09-03:
`multiplexer` is the more accepted spelling

comonoidial, 2016-09-05:
The downside of using this vector multiplexer variant is that it depends on the implicit Enum ordering of the selection signal. And I would prefer explicit selectors to keep the code understandable.

Maybe Vector could be made an instance of Representable Functor, with:
```haskell
tabulate :: (Index n -> a) -> Vec n a
index :: Vec n a ->(Index n -> a)
```
Also the related type classes in Data.Key have a lot of functions that work on both lists and vectors, so that could be a way to simplify the transition from list based code.