clash-lang/clash-prelude #73: Function to replace slice of vector

adamwalker, 2016-09-05:
Would it be possible to add a function, replaceVec, similar to "replace" from CLaSH.Sized.Vector which replaces a slice instead of a single element?
Perhaps something like:

```
replaceVec :: (KnownNat m) => SNat o -> Vec m a -> Vec (o + p + m) a -> Vec (o + p + m) a
replaceVec offset replacement vec = vecTruncated ++ replacement ++ vecTail
    where
    vecTruncated =  take offset vec
    vecTail      =  drop (offset `addSNat` lengthS replacement) vec
```

christiaanb, 2016-09-05:
@adamwalker Do you want a "static" version, like the example you posted (which is sorta similar to [setSlice](http://hackage.haskell.org/package/clash-prelude-0.10.14/docs/CLaSH-Prelude-BitIndex.html#v:setSlice), or do you want a "dynamic" version, similar to the [replace](http://hackage.haskell.org/package/clash-prelude-0.10.14/docs/CLaSH-Sized-Vector.html#v:replace) function you mentioned? Or do you want both?

adamwalker, 2016-09-05:
Good point. I actually want both, if thats doable.

christiaanb, 2016-09-05:
So what should be the out-of-bounds behaviour for the dynamic version?

* Should it be similar to `replace`, and give a run-time error when the `offset + length replacement > length vec`?
* Should it truncate the LSBs of `replacement`?

adamwalker, 2016-09-05:
Well, I think it should behave however you decide is most sensible :)
But, in my opinion it should behave similar to replace and give a runtime error.
What does replace do in actual hardware when the index is out of bounds?

christiaanb, 2016-09-05:
The worst thing possible: the index will wrap around.