clash-lang/clash-prelude #76: Package name "CLaSH" is a disaster to type

firegurafiku, 2016-10-11:
The «two caps — small letter — two caps» is so unusual pattern so I just cannot type it automatically, I need to fix my attention to that ridiculous casing. What's wrong with plain «Clash»?

christiaanb, 2016-10-12:
Well, it all started with "CAES Language for Synchronous Hardware", CLaSH, as the name for the language/compiler. I never gave it much thought when I made the root-name of the module structure the same name as the name of the language/compiler.

tomjaguarpaw, 2016-10-25:
I'd really appreciate changing the top level module to `Clash`.  I know it's an unwieldy change, but better now than later, and it can be done in a backwards compatible way.

christiaanb, 2016-10-25:
I'm inclined to do this for the clash 1.0 release, although I can be convinced to do it for the upcoming clash-0.7 release if people really want it.

expipiplus1, 2016-12-07:
I never had a problem with this, but now every time I type `CLaSH` I think "Oh, I'm experiencing issue #76".

expipiplus1, 2017-01-24:
Also, why is the lambda lowercase if it's standing for "Language"?

christiaanb, 2017-01-24:
because the uppercase lambda isn't as cool

commit 487906e1 (2017-09-15): Rename CLaSH to Clash

Fixes #76
PR #125 (2017-09-15): Rename CLaSH to Clash
Fixes #76
commit 15751f95 (2017-09-15): Rename CLaSH to Clash (#125)

Fixes #76
closed 2017-09-15 by christiaanb