clash-lang/clash-prelude #77: instance BitPack ()

tomjaguarpaw, 2016-10-24:
There should be

````haskell
instance BitPack () where
  type BitSize () = 0
  pack   _ = minBound
  unpack _ = ()
````

I know this seems trivial but it's going to be needed by someone sooner or later (probably because they are generating code).

christiaanb, 2016-10-24:
@tomjaguarpaw what kind of VHDL/Verilog do you expect CLaSH to output given the following Haskell code?:

```haskell
topEntity :: BitVector 0 -> BitVector 0
topEntity = id
```

An entity with 1 input port and 1 output port, both of width 0? Or, an entity without input or output ports?

I'm asking, because VHDL/Verilog synthesis tools loudly complain about things that are 0-bit wide. So if you expect the entity to have an input and an output port, then the instance should be:

```haskell
instance BitPack () where
  type BitSize () = 1
  pack _ = minBound
  unpack _ = ()
```

tomjaguarpaw, 2016-10-24:
I've no idea, but this problem seems to be independent of my suggestion.

tomjaguarpaw, 2016-10-24:
I'm not expecting anyone ever to use this instance in a top-level entity in any case ...