clash-lang/clash-prelude #88: build failure w/ GHC-8.0.2

dmjio, 2017-01-17:
`clash-prelude-0.11` fails to build under ghc-8.0.2 w/
```bash
src/CLaSH/Prelude/Synchronizer.hs:136:17: error:
    • Could not deduce (GHC.TypeLits.KnownNat
                          ((addrSize + 1) - (addrSize - 1)))
        arising from a use of ‘complement’
      from the context: 2 <= addrSize
        bound by the type signature for:
                   isFull :: (2 <= addrSize) =>
                             SNat addrSize
                             -> BitVector (addrSize + 1) -> BitVector (addrSize + 1) -> Bool
        at src/CLaSH/Prelude/Synchronizer.hs:(126,1)-(131,14)
      or from: GHC.TypeLits.KnownNat addrSize
        bound by a pattern with constructor:
                   SNat :: forall (n :: GHC.Types.Nat).
                           GHC.TypeLits.KnownNat n =>
                           SNat n,
                 in an equation for ‘isFull’
        at src/CLaSH/Prelude/Synchronizer.hs:132:17-20
      or from: 1 <= addrSize
        bound by a pattern with constructor:
                   Dict :: forall (a :: Constraint). a => Dict a,
                 in a case alternative
        at src/CLaSH/Prelude/Synchronizer.hs:133:7-10
    • In the first argument of ‘(++#)’, namely
        ‘complement (slice addrSize a1 s_ptr)’
      In the second argument of ‘(==)’, namely
        ‘(complement (slice addrSize a1 s_ptr) ++# slice a2 d0 s_ptr)’
      In the expression:
        ptr == (complement (slice addrSize a1 s_ptr) ++# slice a2 d0 s_ptr)
```

christiaanb, 2017-01-17:
@dmjio what version of the [ghc-typelits-knownnat](http://hackage.haskell.org/package/ghc-typelits-knownnat) package is being used? Because I released version 0.2.3 on 15-Jan-2017 for exactly this bug: http://hackage.haskell.org/package/ghc-typelits-knownnat-0.2.3/changelog

christiaanb, 2017-01-17:
Using stackage LTS-7.16 and GHC-8.0.2 (https://github.com/clash-lang/clash-prelude/blob/0.11/stack-ghc802.yaml), all tests seems to pass: https://travis-ci.org/clash-lang/clash-prelude/jobs/192653376

dmjio, 2017-01-17:
@christiaanb, Ah, that must be it, I'm building under nix. The latest `clash-*` packages aren't on there yet (cabal2nix hackage script has yet to run and commit latest). So I'm cloning locally and overriding, it must be using an old `ghc-typelists-knownat`. Let me try to pull latest on that package.

dmjio, 2017-01-18:
Alright, I've upgrade all packages to latest, included `natnormalise` as well.
```nix
{                                                                                                                                                                                 
  # allowBroken = true;                                                                                                                                                           
  packageOverrides = pkgs: {                                                                                                                                                      
   haskell = pkgs.haskell // {                                                                                                                                                    
    packages = pkgs.haskell.packages // {                                                                                                                                         
      ghc802 = pkgs.haskell.packages.ghc802.override {                                                                                                                            
        overrides = self: super: {                                                                                                                                                
          clash-ghc = self.callPackage ./clash-ghc-0.7/clash-ghc-0.7.nix {};                                                                                                      
          clash-lib = self.callPackage ./clash-lib-0.7/clash-lib-0.7.nix {};                                                                                                      
          clash-prelude = self.callPackage ./clash-prelude-0.11/clash-prelude-0.11.nix {};                                                                                        
          clash-systemverilog = self.callPackage ./clash-systemverilog-0.7/clash-systemverilog-0.7.nix {};                                                                        
          clash-verilog = self.callPackage ./clash-verilog-0.7/clash-verilog-0.7.nix {};                                                                                          
          clash-vhdl = self.callPackage ./clash-vhdl-0.7/clash-vhdl-0.7.nix {};                                                                                                   
          ghc-typelits-knownnat = self.callPackage ./ghc-typelits-knownnat-0.2.3/ghc-typelits-knownnat-0.2.3.nix {};                                                              
          ghc-typelits-natnormalise = self.callPackage ./ghc-typelits-natnormalise-0.5.2/ghc-typelits-natnormalise-0.5.2.nix {};                                                  
         };                                                                                                                                                                       
        };                                                                                                                                                                        
      };                                                                                                                                                                          
    };                                                                                                                                                                            
   };                                                                                                                                                                             
}                                                                                                                                                                                 
```
But am now hitting a new error, unsure if this has anything to do with `clash`.  I can't build packages that depend on `clash-ghc`. If this doesn't look familiar I can file a separate issue, or take it elsewhere.

```bash
Building clash-ghc-0.7...
Preprocessing executable 'clash' for clash-ghc-0.7...

In file included from src-bin/GHCi/UI.hs:33:0: error: 

/nix/store/6n24x08s15sm494zqh6sha1fafhii2v4-ghc-8.0.2/lib/ghc-8.0.2/ghc-8.0.2/include/HsVersions.h:23:0: error:
     fatal error: ../includes/ghcautoconf.h: No such file or directory
     #include "../includes/ghcautoconf.h"
     ^
compilation terminated.
`cc' failed in phase `C pre-processor'. (Exit code: 1)
```

christiaanb, 2017-01-18:
@dmjio: ah, yes, I missed a file in the source release of `clash-ghc-0.7`, you will need to use `clash-ghc-0.7.0.1`.

dmjio, 2017-01-18:
Cool, updated to `0.7.0.1` and I'm in now :) Thanks.

closed 2017-01-18 by dmjio