clash-lang/clash-prelude #96: Changing `()`'s `Bundle` instance to be like the other tuple instances?

christiaanb, 2017-02-28:
Currently we have:

```haskell
instance Bundle () where
  type Unbundled () = Signal ()
  bundle :: Signal () -> Signal ()
  bundle = id
  unbundle :: Signal () -> Signal ()
  unbundle = id
```

which is completely `_|_`-preserving is both directions, _but_, doesn't correspond to the other `Bundle` instances for tuples.

Should this be changed to:

```haskell
instance Bundle () where
  type Unbundled () = ()
  bundle :: () -> Signal ()
  bundle t = pure t
  unbundle :: Signal () -> ()
  unbundle _ = ()
```

which corresponds more to the other `Bundle` instances for tuples, but isn't `_|_`-preserving in its `unbundle` method.

Ericson2314, 2017-03-01:
I've been wishing products were unlifted in Haskell lately. Certainly we need to be lazy about the tag bits, but when there's no tag bits there's no obligation!

Not sure what this means for undefined/X as the bottom we most care about, but otherwise I approve of this change.

christiaanb, 2017-03-01:
The other Product-like instances aren't particularly Bundle class law abiding either when it comes to bottom.

commit 7138907d (2017-03-02): Bundle instance for `()` behaves like a product-type instance

Fixes #96
closed 2017-03-02 by christiaanb