clash-lang/clash-prelude #132: Rename `satMin` to `satSub`

christiaanb, 2017-10-26:
Basically rename:

```
class (Bounded a, Num a) => SaturatingNum a where
  -- | Addition with parametrisable over- and underflow behaviour
  satPlus :: SaturationMode -> a -> a -> a
  -- | Subtraction with parametrisable over- and underflow behaviour
  satMin  :: SaturationMode -> a -> a -> a
  -- | Multiplication with parametrisable over- and underflow behaviour
  satMult :: SaturationMode -> a -> a -> a
```

to 

```
class (Bounded a, Num a) => SaturatingNum a where
  -- | Addition with parametrisable over- and underflow behaviour
  satAdd :: SaturationMode -> a -> a -> a
  -- | Subtraction with parametrisable over- and underflow behaviour
  satSub  :: SaturationMode -> a -> a -> a
  -- | Multiplication with parametrisable over- and underflow behaviour
  satMul :: SaturationMode -> a -> a -> a
```

leonschoorl, 2017-10-26:
If you do this, to keep things consistent you should probably also rename the operations on the [ExtendingNum class](http://hackage.haskell.org/package/clash-prelude-0.11.2/docs/CLaSH-Class-Num.html#t:ExtendingNum) and [bounded{Plus,Min,Mult}](http://hackage.haskell.org/package/clash-prelude-0.11.2/docs/CLaSH-Class-Num.html#v:boundedPlus).

Did you thing about the argument order of sub(tract) vs min(us)?
`boundedMin 3 1` reads like "3 minus 1"
but
`boundedSub 3 1` could be read like "subtract 3 from 1", very different.

martijnbastiaan, 2018-08-28:
@christiaanb Could you elaborate on the reason you want to see this change?