clash-lang/clash-prelude #133: Possible regression with the Traversable instance of Vec

pikkio, 2017-10-27:
Hi, 
I recently moved a project from `lts-3.6` (GHC 7.10.2, clash-ghc 0.5.14,  clash-lib 0.5.12, clash-prelude 0.9.2) to `lts-9.10` (GHC 8.0.2, clash-ghc 0.7.2, clash-lib 0.7.1, clash-prelude 0.11.2) and I suddenly ran into a typing issue when using `Vec`.

```haskell
{-# LANGUAGE DataKinds      #-}
{-# LANGUAGE KindSignatures #-}

import CLaSH.Sized.Vector  (Vec (..))
import GHC.TypeLits        (KnownNat, Nat)
import Prelude

data Foo (n :: Nat) = Foo (Maybe (Vec n Int)) deriving (Show)

foo :: KnownNat n => Maybe (Vec n Int) -> Foo n
foo v = Foo $ v >>= traverse bar

bar :: Int -> Maybe Int
bar x = undefined
```

For example, the (non-synthesizable) code above compiles fine on lts-3.6 (both with `ghc` and `clash`), whereas on lts-9.10 I get:

```
Foo.hs:23:21: error:
    • Couldn't match type ‘n’ with ‘n0 GHC.TypeLits.+ 1’
        arising from a use of ‘traverse’
      ‘n’ is a rigid type variable bound by
        the type signature for:
          foo :: forall (n :: Nat). KnownNat n => Maybe (Vec n Int) -> Foo n
        at Foo.hs:22:8
    • In the second argument of ‘(>>=)’, namely ‘traverse bar’
      In the second argument of ‘($)’, namely ‘v >>= traverse bar’
      In the expression: Foo $ v >>= traverse bar
    • Relevant bindings include
        v :: Maybe (Vec n Int) (bound at Foo.hs:23:5)
        foo :: Maybe (Vec n Int) -> Foo n
          (bound at Foo.hs:23:1)
```

christiaanb, 2017-10-27:
For `clash-prelude-0.10` we changed the `Foldable` (and consequently the `Traversable`) instance so that it only works for non-empty `Vec`tors. Basically, you will have to write `foo` to:

```haskell
foo :: KnownNat n => Maybe (Vec (n + 1) Int) -> Foo (n + 1)
foo v = Foo $ v >>= traverse bar
```

Maybe the following also works.

```haskell
foo :: (KnownNat n, 1 <= n) => Maybe (Vec n Int) -> Foo n
foo v = Foo $ v >>= traverse bar
```

I blame this on `Foldable` being a rediculously large type-class, and including `foldr1` and `foldl1`, which can only be reasonably implemented for non-empty `Vec`tors.

pikkio, 2017-10-27:
Oh, I see. Thank you very much!
The following code compiles perfectly:

```haskell
{-# LANGUAGE DataKinds        #-}
{-# LANGUAGE KindSignatures   #-}
{-# LANGUAGE TypeOperators    #-}
{-# LANGUAGE FlexibleContexts #-}

import CLaSH.Sized.Vector (Vec (..))
import GHC.TypeLits       (KnownNat, Nat, type (+))

foo :: KnownNat (n + 1) => Maybe (Vec (n + 1) Int) -> Foo (n + 1)
foo v = Foo $ v >>= traverse bar

-- etc.
```

closed 2017-10-27 by pikkio